meta/OpenBIC #49: OEM command send_interrupt_to_BMC

Yi-Shum, 2021-11-09:
Hi @rainlinWW,

Quanta is testing ASD function now.
When testing ASD, we need to used OEM command 0x38 0x07 to send GPIO interrupt to BMC , but this command is now used as sensor polling enable.

In YV3, sensor poll enable is 0x38 0x30, and send interrupt to BMC is 0x38 0x07.
Will Wiwynn followed the command used on YV3 in the future?

Thanks,
Eason

rainlinWW, 2021-11-09:
Hi Yi Shum,

The command was temporary set to 0x38 0x7 during previous development.
And yes, we will follow Yv3 commands as possible.

Are you going to implement ASD on Yv3.5 for Craterlake?

Thanks,
Rain

Yi-Shum, 2021-11-09:
Hi @rainlinWW,

Thanks for your answer.
We have implement JTAG command 0x38 0x21, 0x38 0x22 and run BMC ASD csript for testing.
Do you have plans to implement ASD on BIC(command 0x38 0x26&0x28)?

Thanks,
Eason

rainlinWW, 2021-11-09:
Hi Yi Shum,

No, we only schedule to test JTAG signal in current stage.
It will be great if you can help to complete reset of ASD feature on BIC.

Thanks,
Rain

Yi-Shum, 2021-11-09:
Hi rain,

OK, I will push the code about the JTAG command.
In addition, do we need to help implement the interrupt of gpio to send interrupt to BMC during ASD testing?

Thanks,
Eason

rainlinWW, 2021-11-09:
Hi Yi Shum,

Yes, please help to implement the ISR function for sending interrupt to BMC for ASD.
Much appreciate for your help.

Thanks,
Rain

closed 2021-11-09 by Yi-Shum