meta/OpenBIC #50: IPMB RX polling delay and TX sleep time

Yi-Shum, 2021-11-09:
Hi @rainlinWW,

When we run ASD cscript on BMC, it takes more time than YV3, because IPMB TX and RX will sleep 10 milliseconds after sending and receiving a packet.
If we reduce the sleep time, it will improve.

What is the reason for waiting 10 milliseconds on IPMB TX and RX?

Thanks,
Eason

rainlinWW, 2021-11-09:
Hi @Yi-Shum ,

For Rx thread, we suggest to keep the sleep time in that it sets the I2C mqueue polling interval, or there will be frequently context switch impacting performance for all other threads, especially on SKU with both 1ou and 2ou expansion card present.

And for Tx thread, the sleep time is only set to yield the CPU source to other threads, you may try to tune the sleep time for better performance or try replacing the sleep with kernel yield API.

Thanks,
Rain

Yi-Shum, 2021-11-10:
Hi Rain,

Thanks for your answer.
There is another question about IPMB RX, the function ipmb_slave_read() get the latest IPMB packet instead of the oldest.
Is there any reason why ipmb_slave driver uses LIFO?

Thanks,
Eason

rainlinWW, 2021-11-11:
Hi Yi-Shum,

Currently the slave read function only support sending one packet at a time and we don't expect there will be a newer or older packet.

Thanks,
Rain

Yi-Shum, 2021-11-11:
Hi Rain,

Thanks for your answer, but my question is:
If BIC receives two IPMB packets, the first is A and the second is B, the driver will temporarily store these packets in a buffer.
When we call ipmb_slave_read() to get a packet from the driver buffer, and we will get B (the newer one).
Is this the expected action(LIFO)?

Thanks,
Eason

rainlinWW, 2021-11-11:
Hi Yi-Shum,

About API ipmb_slave_read(), it's provided from Aspeed and we are not so sure about the implement.
We will check the code behavior with Aspeed.
Also, could you provide the way you reproduce the issue?

Thanks,
Rain

Yi-Shum, 2021-11-12:
Hi Rain,

We found this issue using ast1030 evb and the following commands.

```
uart:~$ i2c slave_attach IPMB_SLAVE_0
uart:~$ i2c write I2C_1 50 0 01
uart:~$ i2c write I2C_1 50 0 02
uart:~$ i2c write I2C_1 50 0 03
uart:~$ i2c slave_ipmb_read IPMB_SLAVE_0
ipmb length : 3
00000000: a0 00 03                                         |...              |
uart:~$ i2c slave_ipmb_read IPMB_SLAVE_0
ipmb length : 3
00000000: a0 00 02                                         |...              |
uart:~$ i2c slave_ipmb_read IPMB_SLAVE_0
ipmb length : 3
00000000: a0 00 01                                         |...              |
```

Thanks,
Eason

rainlinWW, 2021-11-15:
Hi  Yi-Shum,

Aspeed helped to push commit to modify IPMB slave read to FIFO.
You can try if the issue solved.
https://github.com/AspeedTech-BMC/zephyr/commit/8fcd674716c33c8dba2572a46bc21c622f2c8ade

Thanks,
Rain

Yi-Shum, 2021-11-15:
Hi Rain,

Okay, thanks for your reply.

Thanks,
Eason

closed 2021-11-15 by Yi-Shum