meta/OpenBIC #57: GPIO interrupt trigger fail

Yi-Shum, 2021-11-16:
Hi @rainlinWW,

In irq_callback(), gpio_isr_info is a global variable.
If two GPIO interrupts are triggered at the same time, only one will be added to the work queue.

The GPIO trigger interrupt flow is as follows:
   irq_callback()  ->  gpio_isr_handler()  ->  interrupt callback function defined in plat_gpio_cfg[]
Could we remove the function gpio_isr_handler() and directly add the GPIO callback function into workqueue.

Thanks,
Eason

rainlinWW, 2021-11-17:
Hi Yi-Shum,

Currently we didn't run into such a critical case that two ISR triggered at same moment.
Could you help to provide the way you reproduce the issue?

And yes, please help to remove  gpio_isr_handler() as possible.

Thanks,
Rain

Yi-Shum, 2021-11-17:
Hi Rain,

Thanks for your reply, we will remove gpio_isr_handler().

We have added two GPIO interrupts, PWRGD_CPU_LVC3_R and RST_PLTRST_BUF_N.
During power off, these GPIO pins fall and trigger interrupt at the same time, and this problem is encountered..

Thanks,
Eason

closed 2021-11-17 by Yi-Shum