meta/OpenBIC #90: IPMI get sensor reading

Yi-Shum, 2021-12-21:
Hi @rainlinWW,

The get sensor reading commands (0x04 0x2d, 0x38 0x23) reply completion code 0xC0 (Node Busy) when the sensor is unreadable.
But in YV3, BIC reply completion code 0x00 with three response data 0x00 0x00 0xE0.
Should we follow YV3?

Thanks,
Eason

rainlinWW, 2021-12-22:
Hi @Yi-Shum 

Both completion code and response can return fail status, but we think they should be used for different cases.
According to IPMI specification, software should be able to avoid reading unreadable sensors continuously as possible, and should return 0x00 0x00 0xE0 in expected corner case like trying to access sensor at the moment powering off.
So in OEM accurate sensor reading, we returned 0x00 0x00 0xE0 as BIC checking the sensor shouldn't be access at the moment.
And return corresponding completion code as BIC detecting error.

Do you think it's better to do this way, or is there any reason you think we should follow Yv3?

Thanks,
Rain

Yi-Shum, 2021-12-23:
Hi @rainlinWW,

Thanks for your reply.

We have an issue about sensor reading.
If only insert one DIMM, BIC will return completion code 0xC0 when BMC read other DIMMs which is not inserted.
This causes BMC retry to read these DIMMs.
Is it expected?

Thanks,
Eason

rainlinWW, 2021-12-23:
Hi @Yi-Shum 

In Yv3, BIOS will notify BMC/BIC about DIMM present status and only poll present DIMMs.
We will work with BIOS team to solve the issue.

Thanks,
Rain

Yi-Shum, 2021-12-23:
Hi @rainlinWW,

got it

Thanks,
Eason

Yi-Shum, 2021-12-23:
Hi @rainlinWW,

We checked the behavior in Yv3, but it is different from what you said.
Please refer to the following:
```
root@bmc-oob:~# bic-util slot1 0x10 0x2d 0x6
24 C0 C0

(the following is to read not present DIMMs)
root@bmc-oob:~# bic-util slot1 0x10 0x2d 0x7
00 E0 C0
root@bmc-oob:~# bic-util slot1 0x10 0x2d 0x9
00 E0 C0
```

By the way, the SSD0 Temp also has this problem.

Thanks,
Eason

rainlinWW, 2021-12-24:
Hi @Yi-Shum 

Currently we only use OEM accurate sensor reading for sensor polling from BMC.
Please check the command behavior.

Thanks,
Rain

GoldenBug, 2022-01-13:
Hi @Yi-Shum,

Has this issue been resolved or is it still ongoing?

Yi-Shum, 2022-01-13:
Hi @rainlinWW,

Using OEM accurate sensor reading command to read temperature of DIMMs not inserted, BIC returns the completion code 0xC0.
Sensor polling with accurate sensor reading command also causes BMC retry to read these DIMMs.

Thanks,
Yi-Shum

rainlinWW, 2022-01-18:
Hi @Yi-Shum 

BIC cannot detect DIMM present status, so we think BIC should not request BMC to stop polling DIMM temperature due to BIC fail to read DIMM temperature from CPU.
This issue was once solved in Yv3 and the solution relied on BIOS to report DIMM present status to BMC.
Please help to close this issue if there is no further questions.

Thanks,
Rain

Yi-Shum, 2022-01-19:
Hi @rainlinWW,

Thanks for your reply.
Yi-Shum

closed 2022-01-19 by Yi-Shum