meta/OpenBIC #98: Event data of FRB3/CATERR

Yi-Shum, 2021-12-27:
Hi @rainlinWW,

The value of FRB3, IERR and MCERR is different from Yv3 in sdr.h.
Should we modify it to be the same as Yv3?

GPIO        | Yv3  | Yv3.5 
--------------|-------|------
IPMI_SDR_SENSOR_SPEC_EVENT_PROCESSOR_IERR    | 0x00 |  0x01 
IPMI_SDR_SENSOR_SPEC_EVENT_PROCESSOR_FRB3    | 0x04 |  0x10
IPMI_SDR_SENSOR_SPEC_EVENT_PROCESSOR_MCERR  | 0x0B |   0x08

Thanks,
Eason

rainlinWW, 2021-12-29:
Hi @Yi-Shum 

Definition in common/sensor/sdr.h is common definition for all projects.
We don't expect it to be modified for specific project.

Thanks,
Rain

Yi-Shum, 2021-12-29:
Hi @rainlinWW,

But according to the IPMI specification "42.2 Sensor Type Codes and Data", the definition of event data is same as Yv3.

Thanks,
Yi-Shum

Yi-Shum, 2021-12-29:
Hi @rainlinWW,

Thanks for your reply.
I was wrong to think the define in sdr.h is used as event data.
The event data is defined in libipmi.h at Yv3, we would copy this file to here and add GPIO interrupt to add FRB3/CATERR event log. 

Thanks,
Yi-Shum

closed 2021-12-29 by Yi-Shum