meta/OpenBIC #102: Questions about IPMB

Yi-Shum, 2021-12-28:
Hi @rainlinWW,

We have some questions about IPMB design.
1. In function ipmb_send_response() and ipmb_send_request(), the waiting time to put message into queue is K_forever.
Should it set a appropriate timeout value?
2. The queue size of ipmb_txqueue and ipmb_rxqueue is 1 and 2.
Is there any reason to set this size?
3. In ipmb_read(), the mutex lock and mutex unlock function doesn't check the return value.

Thanks,
Yi-Shum

rainlinWW, 2021-12-29:
Hi @Yi-Shum 

1. We will have a discuss about the timeout value.
2. The tx and rx queue are for message passing between threads only and tx thread polling frequency should be fast enough to handle messages in time. Due to memory size limit in BIC, we would like to save memory size as possible.
DO you think or meet any issue cause by not enough queue size?
3. Yes, there should be a mutex status check.

Thanks,
Rain

Yi-Shum, 2021-12-30:
Hi @rainlinWW,

Thanks for your reply.
The rx queue only be get in function ipmb_read(). Because it would be purge before get, should the size of rx queue also be 1? 

There is another question about ipmb_read().
In ipmb_read(), the mutex(mutex_read) used for all IPMB bus is same. Is it expected?

Thanks,
Yi-Shum

closed 2022-03-09 by Yi-Shum