meta/OpenBIC #112: Questions about PR #108

rainlinWW, 2022-01-06:
Hi @Yi-Shum 

In PR #108 , is there any issue about malloc ipmi_msg so you have to modify it to declare directly instead?
From our point of view, malloc big size variable like ipmi_msg can reduce memory usage in thread stack, which means we can reduce size of all threads calling this function.

Also, for events that requires delay, suggest using k_work instead of creating extra thread and occupy kernel memory.

Yi-Shum, 2022-01-06:
Hi @rainlinWW,

Thanks for your suggestion.

We directly declare ipmi_msg to avoid memory leaks, because threads proc_fail_thread_handler and CatErr_thread_handler may be aborted.

We will modify these two thread to use the timer instead of thread, and restore to dynamic allocation ipmi_msg.

Thanks,
Yi-Shum

GoldenBug, 2022-01-13:
Hi @Yi-Shum,

Has this change been made and merged yet or is it still ongoing?

Yi-Shum, 2022-01-13:
fby3.5: cl: Modify FRB3 and CATERR event log (#117) solve this issue.

Yi-Shum, 2022-01-13:
Hi @GoldenBug,

It has been merged.

Thanks,
Yi-Shum

rainlinWW, 2022-01-18:
Hi @GoldenBug 

Both of my concerns are fixed.
Please help to close this issue if no further questions.

Thanks,
Rain

closed 2022-01-20 by GoldenBug