meta/OpenBIC #116: The build steps/configs

YoungLi123, 2022-01-07:
Hi @rainlinWW,

We found that the released firmware (xxxx.bin) size is much different with UART firmware (uart_xxxx.bin),

For example:
oby35-cl-2022.01.01.bin about 321KB
uart_oby35-cl-2022.01.01.bin about 261KB

We follow https://github.com/facebook/OpenBIC#readme to build.

Both xxxx.bin and uart_xxxx.bin size are almost the same (about 261KB).

Is there any steps/configs we miss?

We just want to make sure our build steps is OK.

Thanks,
Young

rainlinWW, 2022-01-07:
Hi Young,

The size different is due to extra data for formal release firmware secure update.
Build steps in openbic readme file is fine for firmware development.

Thanks,
Rain

YoungLi123, 2022-01-07:
Got it!

Thanks,
Young

closed 2022-01-07 by YoungLi123