meta/OpenBIC #143: BIC cold-reboot issue

MouchenHung-QUANTA, 2022-01-26:
Hi @rainlinWW,

We recently encountered an BIC reboot issue. Host would hang-up while sending cold-reboot command(0x06,0x02). Have you seen this problem before?

Thanks,
Mouchen

rainlinWW, 2022-01-26:
Hi @MouchenHung-QUANTA 

It's a known issue that BIC will fail to bypass host console to BMC after cold reset.
Could you help to check if host ssh still works after cold reset on your side?

Thanks,
Rain

MouchenHung-QUANTA, 2022-01-26:
Hi @rainlinWW,

Well, host still works after cold reset command via ssh method. That means no problem with host.
So, does this known issue already got solution?

Thanks,
Mouchen

rainlinWW, 2022-01-27:
Hi @MouchenHung-QUANTA 

Currently we are working with Aspeed to find root cause.

Thanks,
Rain

LoraLin1, 2022-06-14:
Hi @MouchenHung-QUANTA 

Aspeed system warm reset default setting is SOC reset, and system cold reset is full chip reset.
BIC would do full chip reset before modifying cold reset command reset type.
Full chip reset would cause SUART disabled between BIC and host, and BMC can't connect to host through SOL after cold reset.

We modify cold reset command reset type from full chip reset to SOC reset.
https://github.com/facebook/OpenBIC/commit/34644ded2e8d0bfa56c393648fdf45fb177064e4

closed 2022-06-27 by MouchenHung-QUANTA