meta/OpenBIC #154: IPMB_TXTask wait forever when bridge message from host fail

Yi-Shum, 2022-02-09:
Hi @rainlinWW,

We have an issue about IPMB_TXTask.
If BMC and host reboot at the same time, the bridge message from host will fail with the maximum number of retries.
In IPMB_TXTask, it will respond CC_NODE_BUSY to KCS and wait forever in function ipmb_send_response().

Thanks,
Yi-Shum

Yi-Shum, 2022-02-15:
Hi @rainlinWW,

We will add an else if statement to the IPMB_TXTask to print an error message when the bridge message from the host reaches the maximum number of retries.

Do you have any suggestions for this modification?
If not, we will submit this PR.

Thanks,
Yi-Shum

DelphineChiu, 2022-02-15:
Hi @Yi-Shum  Rain is taking one day off today. you can go ahead with your solution. Thanks.

Yi-Shum, 2022-02-15:
Hi @DelphineChiu,

OK, thanks for your reply.

Yi-Shum

DelphineChiu, 2022-02-16:
Hi @Yi-Shum 

Sorry that we do mode handle for this case in https://github.com/facebook/OpenBIC/pull/169 and also includes the error messages. Do you feel good to close https://github.com/facebook/OpenBIC/pull/165 ?

Yi-Shum, 2022-02-16:
Hi @DelphineChiu ,

OK, I will close it

Yi-Shum, 2022-02-17:
#169 solve this issue