meta/OpenBIC #161: Add OEM platform command

MouchenHung-QUANTA, 2022-02-10:
Hi @rainlinWW,

In order to debug or get some server's info while BIC running, we have a suggestion to add an OEM platform command in BIC console which may includes GPIO get/set or other topics in further.
Is that ok for you?

p.s. Bellow is one of the example from current work.
 
[image]

Thanks,
Mouchen

rainlinWW, 2022-02-10:
Hi @MouchenHung-QUANTA 

We think it a great command and very useful during development.
But there is a concern that Zephyr doesn't support reading GPIO direction and it might be risky to return recorded status in GPIO table.

Thanks,
Rain

MouchenHung-QUANTA, 2022-02-10:
Hi @rainlinWW,

Indeed, zephyr-aspeed gpio driver doesn't support direction get/set api, so I used GPIO table's (gpio_cfg) direction instead temporary to get initial sets. Even though, this could be discussed or modified in the future.

Thanks,
Mouchen

MouchenHung-QUANTA, 2022-02-10:
Hi @rainlinWW,

According to your doubt, we thought that getting direction from the table may not cause serious problem. If there's any way to get current gpio direction in the future, we will update table synchronously.

Thanks,
Mouchen

rainlinWW, 2022-02-14:
Hi @MouchenHung-QUANTA 

Ok, thanks for your help.

Thanks,
Rain

closed 2022-03-10 by MouchenHung-QUANTA