meta/OpenBIC #182: The frequency setting of I2C for standard mode is faster than the criteria

YoungLi123, 2022-02-25:
Hi @DelphineChiu

Our EA team measure the I2C bus signal and found that the frequency of I2C[2] (sensor bus) and I2C[5] (VR bus) is 100.8 KHz.

It's faster than the maximum speed of standard mode (100KHz).

Do you also see the symptom and do we have a plan to decrease the frequency setting?

Thanks
Young

BonnieLo, 2022-05-10:
Hi @YoungLi123 

We have discussed the I2C frequency issue with Aspeed and they share their change with us as follows:

" Please help to sync i2c driver from that tag.
  https://github.com/AspeedTech-BMC/zephyr/blob/v00.01.05/drivers/i2c/i2c_aspeed.c
  https://github.com/AspeedTech-BMC/zephyr/blob/v00.01.05/drivers/i2c/i2c_global_aspeed.c"
  
 We have done the test and it works. Those changes plan to be added in Aspeed Zephyr 00.01.06.
 
Thanks,
Bonnie

YoungLi123, 2022-06-29:
Change Freq from 100.8KHz to 99.21KHz

closed 2022-06-29 by YoungLi123